elliotwoods/ofxKinectForWindows2 #22: Error initializing BodySource (Fix Included)

rickbarraza, 2015-02-28:
ENVIRONMENT:
Working with the OF 0.8.4 tagged branch of the ofxKinectForWindows2 addon, with Visual Studio 2012 Express and the current, official release of OF 0.8.4 as defined here: http://rbarraza.com/setting-up-kinect-2-for-openframeworks/

ISSUE:
Adding "kinect.initBodySource();" in your ofApp.cpp > setup() function causes an ACCESS VIOLATION error in "ofxKinectForWindows2\Source\Body.cpp" line 120:

120:    if (FAILED(frame->GetAndRefreshBodyData(_countof(ppBodies), ppBodies))){
121:		throw Exception("Failed to refresh body data");
122:	    }

SOLUTION:
By comparing the master branch, you'll see that ppBodies isn't properly initialized before being called. So simply add the definition before calling it like so:

120:    IBody* ppBodies[BODY_COUNT] = {0};  // <-- this is the fix.
121:    if (FAILED(frame->GetAndRefreshBodyData(_countof(ppBodies), ppBodies))){
122:          throw Exception("Failed to refresh body data");
123:	   }

And you're good to go.

Let me know when you update, I'll be posting a skeletal tutorial next.

elliotwoods, 2015-02-28:
ok i'll have a look at this today

aha yes
this fix is already in 0.9.0:
https://github.com/elliotwoods/ofxKinectForWindows2/blob/master/src/ofxKinectForWindows2/Source/Body.cpp#L75

i'm tempted to combine both versions (using ifdefs to handle the different syntax between 0.8.4 and 0.9.0 rather than different tags)

will fix today

elliotwoods, 2015-02-28:
hey!

fixed:
https://github.com/elliotwoods/ofxKinectForWindows2/tree/0.8.4

WARNING:
the tag '0.8.4' is now removed. and replaced by branch 0.8.4
(tags aren't designed to be updated).

Please update the tutorial to link to branch 0.8.4 rather than tag 0.8.4

elliotwoods, 2015-02-28:
Ah ok, the URL you linked to in your tutorial now goes to the branch anyway (rather than tag)
So just the language should say 'branch' instead of 'tag' (but thankfully it's less urgent)

closed 2015-02-28 by elliotwoods