elliotwoods/ofxKinectForWindows2 #23: Body.cpp issue

liquidzym, 2015-03-06:
hey elliot,sorry to post here,i just download this addons
but something saying warning MSB8027: Two or more files with the name of Body.cpp will produce outputs to the same location. This can lead to an incorrect build result.  The files involved are ..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Data\Body.cpp, ..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Source\Body.cpp.

and casuing soemthing like this
error LNK2001: 无法解析的外部符号 "public: virtual class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __thiscall ofxKinectForWindows2::Source::Body::getTypeName(void)const " (?getTypeName@Body@Source@ofxKinectForWindows2@@UBE?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)

do you have any ideas?
cheers
alex

elliotwoods, 2015-03-06:
Is this with the examples or with your own project?

Visual Studio creates an obj file for every cpp file it builds. All the obj files for one project are put in the same folder (regardless of the source path). In your case my guess is that you have the addon and another 'body.cpp' in the same project. Both files are being built to Body.obj and they are conflicting

This usually isn't a problem as the addon is kept in a separate project than your app project, for details on how to do that check the readme or http://rbarraza.com/adding-ofxkinectforwindows2-to-a-new-or-existing-of-project/

elliotwoods, 2015-03-06:
Hope that helps. Would be happy to catch up on Skype otherwise (been hoping to have a chat with you sometime!)

liquidzym, 2015-03-07:
i'm running the default example actually, there're two Body.cpp source one is in Source folder,another one in Data folder. not too sure about this,i'm using the master branch.
sure,be happy to talk to you on Skype somthime

elliotwoods, 2015-03-07:
ah yes, good point (Body is in data too)
as a quick fix, rename the `Body.cpp` in the data folder to `Body_.cpp`

liquidzym, 2015-03-08:
ok,that works,thxs!

closed 2015-03-08 by liquidzym