elliotwoods/ofxKinectForWindows2 #27: Create a new project with ofxKinectForWindows2 - building fails

Vicky3, 2015-03-12:
Hi!

I hope this is the right place to put my problem/question.

Yesterday I've downloaded the master-branch of oF and the master branch of ofxKFW2 and the examples run well (at least the first two examples; for the third one I had to download too many packages for just having a look).
Today I wanted to start my own project using ofxKFW2 but I did not succeed in getting a buildable project.

I've tried adding ofxKFW2 to a project as decribed in the ReadME (Hoping that Rightclick on Solution -> Add -> Existing Project does step 1.)
Then I got:
> Error	21	error C1083: Cannot open include file: 'ofxKinectForWindows2.h': No such file or directory	c:\openframeworks\openframeworks-master\apps\myapps\cameralensoffsetexamplecopy\src\ofApp.h	5	1	cameraLensOffsetExample
> Error	24	error C1083: Cannot open include file: 'ofxKinectForWindows2.h': No such file or directory	c:\openframeworks\openframeworks-master\apps\myapps\cameralensoffsetexamplecopy\src\ofApp.h	5	1	cameraLensOffsetExample
>	25	IntelliSense: cannot open source file "ofxKinectForWindows2.h"	c:\openFrameworks\openFrameworks-master\apps\myApps\cameraLensOffsetExampleCopy\src\ofApp.h	5	1	cameraLensOffsetExample
>	26	IntelliSense: name followed by '::' must be a class or namespace name	c:\openFrameworks\openFrameworks-master\apps\myApps\cameraLensOffsetExampleCopy\src\ofApp.h	43	3	cameraLensOffsetExample

Then I tried using the projectGenerator. I also had to add the ofxKinectForWindows2Lib.vcxproj, but now the error is different:
> Error	2	error LNK2001: unresolved external symbol _GetDefaultKinectSensor@4	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj	mergeOffsetExAndKinect
> Error	3	error LNK2001: unresolved external symbol "public: virtual class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __thiscall ofxKinectForWindows2::Source::Body::getTypeName(void)const " (?getTypeName@Body@Source@ofxKinectForWindows2@@UBE?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj	mergeOffsetExAndKinect
> Error	4	error LNK2001: unresolved external symbol "public: virtual void __thiscall ofxKinectForWindows2::Source::Body::init(struct IKinectSensor *)" (?init@Body@Source@ofxKinectForWindows2@@UAEXPAUIKinectSensor@@@Z)	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj	mergeOffsetExAndKinect
> Error	5	error LNK2001: unresolved external symbol "public: virtual void __thiscall ofxKinectForWindows2::Source::Body::update(void)" (?update@Body@Source@ofxKinectForWindows2@@UAEXXZ)	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj	mergeOffsetExAndKinect
> Error	6	error LNK2001: unresolved external symbol "public: void __thiscall ofxKinectForWindows2::Source::Body::drawWorld(void)" (?drawWorld@Body@Source@ofxKinectForWindows2@@QAEXXZ)	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj	mergeOffsetExAndKinect
> Error	7	error LNK2001: unresolved external symbol "public: class ofMatrix4x4 __thiscall ofxKinectForWindows2::Source::Body::getFloorTransform(void)" (?getFloorTransform@Body@Source@ofxKinectForWindows2@@QAE?AVofMatrix4x4@@XZ)	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj	mergeOffsetExAndKinect
> Error	8	error LNK1120: 6 unresolved externals	C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\bin\mergeOffsetExAndKinect.exe	mergeOffsetExAndKinect

I'm using Win8.1 and VS2012. Maybe I made a stupid mistake because I'm not really familiar with VS, but with other Addons I do not have problems.

Can you help me?

Cheers,
Vicky

elliotwoods, 2015-03-12:
> Error 3 error LNK2001: unresolved external symbol "public: virtual class std::basic_string,class std::allocator > __thiscall ofxKinectForWindows2::Source::Body::getTypeName(void)const " (?getTypeName@Body@Source@ofxKinectForWindows2@@UBE?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) C:\openFrameworks\openFrameworks-master\apps\myApps\mergeOffsetExAndKinect\Device.obj mergeOffsetExAndKinect

Hi Vicky 

Its fine to post questions here. I think maybe the official place is the forum, using @elliotwoods to hail me (so other users with the same issue can read along more easily). But no worries.

From the readme :

Usage

Add the ofxKinectForWindows2Lib.vcxproj to your solution
In Property Manager, right click on your project to select Add Existing Property Sheet... and select the ofxKinectForWindows2.props file

It sounds like you missed the step 2 there (the property sheets).
Also, there's a step 3 which is missing from the readme for some reason.

But this guide might help :
http://rbarraza.com/adding-ofxkinectforwindows2-to-a-new-or-existing-of-project/

Its much more in-depth than the readme.

Elliot

Vicky3, 2015-03-13:
Thanks a lot!
I've successfully added the addon to my project! I really don't know why it fails yesterday because I made all these steps (maybe in a wrong order - I didn't know what I'm doing, just tried out several things).
I still fail in creating a new project with the ProjectGenerator, even if I follow the other tutorial by rickbarraza, but it's ok since I can add ofxKFW2 after setting up the project.

elliotwoods, 2015-03-13:
just to note : rick made 2 tutorials
one adding the addon with project generator
the other using my method (second tutorial, the one i linked)

closed 2015-08-15 by elliotwoods