elliotwoods/ofxKinectForWindows2 #29: Examples

cbtriplec, 2015-03-19:
Just wondering if the examples for OF 0.8.4 were working when when you switched to working with OF0.9.0?  I'm running into several issues while building them with 0.8.4 

error C2733: 'GetProcessInformation' : second C linkage of overloaded function not allowed
error LNK1104: cannot open file ...\of_v0.8.4_vs_release\addons\ofxKinectForWindows2\example\bin\ofxKinectForWindows2Lib.lib'

elliotwoods, 2015-03-20:
Are you using the 0.8.4 branch?
I'm using that and the examples work for me last time I checked.
A specific example doesn't work for you? Did you follow Rock Baraza's tutorial?

elliotwoods, 2015-03-20:
Sorry closed by accident

cbtriplec, 2015-03-20:
Yep, using 0.8.4 branch right now.  After looking at the closed issues it looks like that you've mentioned its a mismatch between 64 bit OF and 32 bit (https://github.com/elliotwoods/ofxKinectForWindows2/issues/10) 

I'm pulling 64 bit OF from https://github.com/openframeworks/openFrameworks/tree/feature-64bit is this correct or should i be looking elsewhere? 

I've followed http://rbarraza.com/adding-ofxkinectforwindows2-to-a-new-or-existing-of-project/ tutorial but I'll give it another shot - might have missed something.

cbtriplec, 2015-03-20:
Looks like Rock Baraza's goes over those errors and comments a few things out... my apologies. I should be able to get things working.  Thanks again for your work on this project.. hopefully I don't have to bug you again.

closed 2015-03-20 by cbtriplec

elliotwoods, 2015-03-20:
Make sure to check rick's second tutorial. That highlights the method which I'm using 
Let me know how it goes