elliotwoods/ofxKinectForWindows2 #30: How to map depth point to camera space

vertgo, 2015-05-12:
I see that the depth has a protected coordinate mapper which I think wraps microsoft's coordinate mapper. I also see that in the sample for drawing bodies, it draws the rgb and the depth as a mesh. But the depth is a mesh of depth resolution, 512x424, while the RGB is 1920x1080, and it's offset along the X. 

How do I use the coordinateMapper to call MapDepthPointToCameraSpace so I can draw the pixels RGB pixels directly on their depth point? Right now it's just binding the 1080p texture to the lower resolution mesh and calling it a day.

```c++
kinect.getColorSource()->getTexture().bind(2);

ofSetColor(255);
ofMesh mesh = kinect.getDepthSource()->getMesh(true, ofxKFW2::Source::Depth::PointCloudOptions::ColorCamera);
mesh.draw();
kinect.getColorSource()->getTexture().unbind(2);
```

Is there a better way to get the right colors onto the right vertices?

elliotwoods, 2015-08-15:
maybe the line 

```c++
			this->coordinateMapper->MapDepthFrameToCameraSpace(frameSize, this->pixels.getPixels(), frameSize, (CameraSpacePoint*) mesh.getVerticesPointer());
```

in Depth.cpp can help to suggest how to use the coordinate mapper directly (rather than using the implementation inside ofxKinectForWindows2

elliotwoods, 2015-12-04:
i think i've answered this one so closing

closed 2015-12-04 by elliotwoods