elliotwoods/ofxKinectForWindows2 #33: Can't find ofxAddonLib.props when using Example with GUI in 0.84 branch

Taboen, 2015-07-10:
When trying to load the exampleWithGui/example.sln file in the 0.84 branch two of the four projects fail:
example.vcxproj en ofxCvGuiLib.vcxproj

When loading example.vcxproj the error indicates that it can't find ofxCvGui\ofxCvGuiLib\ofxCvGuiLib.props, even though I changed example.vcxproj imports to refer to ofxCvGui2\ofxCvGuiLib\ofxCvGuiLib.props, so I don't know why it still looks in the ofxCvGui\ folder.

When loading ofxCvGuiLib.vcxproj it tells me it can't find ofxAddonLib.props in libs\ofxAddonLib. When I search on google for this ofxAddonLib it doesn't seem to exist (anymore). 

The other example without GUI works fine.

elliotwoods, 2015-07-10:
Ah sorry
It's new
Maybe I didn't upload
Will do when back at desk

elliotwoods, 2015-07-12:
It's uploaded now
https://github.com/elliotwoods/ofxAddonslib

closed 2015-07-12 by elliotwoods