elliotwoods/ofxKinectForWindows2 #34: Application goes to 20 fps when running kinect.update() - threading breaks entire application

Taboen, 2015-07-11:
So, using this plug in of course I call kinect.update() every frame. If I do this the framerate drops to 20fps. 
Because I want to make it faster, I thought I'd thread the thing. This doesn't work. Calling ofThread::startThread() oddly blanks the GUI and the application reports nothing. 

Some code:

class KinectThread : public ofThread 
{ 
public:
    // the thread function
    void threadedFunction() {
 
        // start
 
        while(isThreadRunning()) {
			lock();
			kinect.update();
			unlock();
		}
 
        // done
    }
 
    ofxKFW2::Device	kinect;
};

setup code:
	kinectThread.kinect.open();
	kinectThread.kinect.initDepthSource();
	kinectThread.kinect.initColorSource();
	kinectThread.kinect.initInfraredSource();
	kinectThread.kinect.initBodySource();
	kinectThread.kinect.initBodyIndexSource();

	kinectThread.startThread();

If I don't call kinectThread.startThread() but instead call kinectThread.kinect.update() every frame it does work as it did before (20fps). 

What's happening? Can I not thread it? Is there a way to make it faster?

elliotwoods, 2015-07-12:
A few things:

You are calling all the setups in one thread and the update in a different thread. Usually with devices, you need to call all operations in the same thread

Update needs to be called in the main thread if you want to use any of the textures (as textures can only be uploaded in the main thread in oF's opengl)

Probably threading won't increase speed (if update is really slow). Since you are going to lock and wait for update to happen in the thread. Threading is never a magic bullet for speed in all situations, it's generally a magic bullet to make more than one thing happen simultaneously where doing so makes sense

General slow down reasons:
1. You are running in debug mode rather than release
2. There is interference on the kinect cable (I get this a lot, it slows the whole app down)
3. Your computer is too slow (I generally find it fine on i5/i7)

elliotwoods, 2015-07-12:
what's your system? do you get kinect performance issues elsewhere?

Taboen, 2015-07-14:
Thanks for your advice.
So I played around a bit. I discovered that my application is a lot faster if I only run the BodyIndexSource rather than all of them, BodyIndexSource is the only I need in the end probably, so that's good. Good to know that threading does not make sense in this situation. It runs at 50fps in this case. 
I checked some other samples and in those the FPS is very similar to that in my application, al though I can not see the FPS directly in the samples. 
My system is a laptop, with i7 @ 2Ghz, 8GB RAM, Geforce GT 650M.

elliotwoods, 2015-07-15:
i've been finding a lot of problems recently with 'bad signal' causing severe slow downs. but it varies from computer-to-computer a lot (e.g. my Mac Pro is really temperamental)

threading can make sense if you want to do something at the same time as using the kinect.

also i think there might be a more clever way of dealing with the update so that the app fps isn't affected in general (e.g. an `isFrameNew`) pattern

elliotwoods, 2015-07-15:
good to hear that the bit you need is running ok for you :)

closed 2015-07-15 by elliotwoods

Taboen, 2015-07-15:
I wanted to create a more clever way of dealing with the update, so I've looked at the code, but I was unable to find a way to only run an update() when a new frame is available. I've tried to use SubscribeFrameArrived(), as part of the reader, but I was unable to use the WAITABLE_HANDLE that it uses. I tried several ways to make it work but none did. It's not important though because 50fps is good enough for me. It was mainly curiosity that was driving me :)