elliotwoods/ofxKinectForWindows2 #35: Unable to build on Win10 & VS2012

braitsch, 2015-08-12:
Hi Elliot,

I'm getting a weird error when trying to build your addon (branch 0.8.4).

```
error C2733: 'GetProcessInformation' : second C linkage of overloaded function not allowed	c:\users\braitsch\code\of_v0.8.4\libs\quicktime\include\processes.h	195	1	ofxKinectForWindows2Lib
```
Have you seen this before and if so any idea on how to fix?

Cheers,
Stephen

elliotwoods, 2015-08-12:
It might be related to your include order
Try putting the ofxKinectForWindows.h above or below your ofMain include

braitsch, 2015-08-12:
Yeah tried switching the order of the include statements and it stills failed to build. Looks like it's a more of a [general OF issue as discussed here](http://forum.openframeworks.cc/t/error-c2733-second-c-linkage-of-overloaded-function-after-moving-to-of-0-8-0/13333/6). I commented out the offending line and now it builds fine.

closed 2015-08-12 by braitsch

elliotwoods, 2015-08-12:
Note that QuickTime might not work now though in the video player