elliotwoods/ofxKinectForWindows2 #36: Issues compiling

mattfelsen, 2015-08-28:
Howdy. I've been having trouble getting things to compile recently (both examples & new/clean projects). Before I go too much further with digging into things and reporting specific errors, I wanted to get a sense of what the latest known working configurations are. The only project I could get running was the basic example on OF 0.8.4, ofxKFW2 on the 0.8.4 branch, VS2012, Windows 8.1, Win32. Beyond that, I haven't had success with any combination of VS (2012/2015), OF (current master from github), architecture (Win32/x64), config (Debug/Release), various example projects...I think that's all the variables :smile: 

So, can you provide a breakdown of what's worked for you?
- Windows version
- VS version
- OF commit 
- ofxKFW2 commit 
- architecture
- config

Pinging a couple others as well @prisonerjohn @robotconscience Thanks so much!

robotconscience, 2015-08-28:
Hey Matt,

I'm using latest commit, Windows 8, latest OF (from git), 64-bit arch, visual studio 2015. Literally just made a new project yesterday!

If you're in VS 2k15, the trickiest thing is adding the reference to the ofxK4W2 product (it's now in its own menu). Once you've the project setup (added ofxK4W2 project, added property sheet), from the Project file menu, go down to "Add Reference", check the box next to "ofxKinectForWindows2Lib", and you should be golden.

mattfelsen, 2015-08-28:
Thanks Brett! I was hoping someone had that setup working as it makes the most sense.

That "Add Reference" step worked! Not at first, which is weird (I was still getting the same errors at first, which were unrelated to linking), but I think I'm good now after creating yet another new project. For reference, I was doing Solution Explorer > Solution > Properties > Common Properties > Project Dependencies > ofxKinectForWindows2Lib > Check, though that seems to be slightly different than what you suggested. Anyhow, in a new project this got me squared away to start drawing streams on screen, so I think that's squared away.

The example & example-bodyIndex projects are still giving me grief, but if I can get the code running from a clean project maybe it's an issue with the project files & I can send along a PR

mattfelsen, 2015-09-12:
Finally got around to investigating this a little more. I've been up & running with newly created projects after removing the LongExposureInfrared class, but wanted to figure out what's going on here. With the example project I was getting a variety of errors (I think x64/Debug compiled though), so I created a new project with the commandLine pg and followed [Rick Barazza's guide](http://rbarraza.com/adding-ofxkinectforwindows2-to-a-new-or-existing-of-project/) without selecting the addon in the pg and adding the ofxKFW2Lib project & props file to the solution after.

On this new project, every configuration *except* x64/Release works. That config gives:

```
C1083 Cannot open compiler generated file: 'x64\Release\\Users\mattfelsen\Documents\Code\openFrameworks\github\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Source\LongExposureInfraRed.obj':
No such file or directory
```

The path to the .obj file is clearly messed up, but it seems like only LongExposureInfrared is affected. I can remove those .h/.cpp files from the project and it compiles fine, and adding them back breaks it again. I've looked at the props, vcxproj, and vcxproj.filters files and don't see anything that looks amiss or different from any of the other files in the ofxKFW2Lib project. Really strange?

elliotwoods, 2015-09-13:
Memo had the same problem

He found out that the issue was because the path was too long
Try moving openFrameworks to a shorter path (e.g. C:\openFrameworks)

See if that fixes things and go from there

mattfelsen, 2015-09-13:
Wow, that seems...ridiculous. Good to go after moving the directory up a couple levels :smiley:

PR #37 (2015-09-17): Fix example project, add new example for BodyIndex
Hi! Two commits here (I kept them separate in case you want to cherry-pick)..

The first is a minor change to example.sln. I noticed that it's trying to build OF in Win32 even when the solution is set to x64, which was causing some of the issues I saw before, and should fix #36 outside of the path-too-long issue :smile:.

The other is the addition of a new example which shows how to pull data from the color image to match the body index image for a green screen effect. I tried to heavily comment it & make the code straightforward so it's appro [... cut]
closed 2015-09-17 by elliotwoods