elliotwoods/ofxKinectForWindows2 #39: Opening a specific kinect if multiple connected

usama-ghufran, 2015-10-21:
In the ofxkinect addon you can open a specific kinect V1 by either the serial id or index in the kinect.open() function. Do we have any way we can access a specific kinect V2 if we have multiple connected. 
I have 2 kinect V2 and both are connected to two different USB 3.0 ports however I have only one controller. I assume i might not be able to open both simultaneously but i should be able to switch between them. Currently with open() it opens the first kinect connected.

usama-ghufran, 2015-10-24:
I've been reading forums and it feels it's a issue with the Kinect2.0 SDK from microsoft. currently they don't seem to have a way to access a kinect by id. it'll open the 'default' kinect device only. :/
The reason they give is that multiple kinect v2 cannot be attached to a single PC due to bandwidth issue. 
Hmm, looks like i'll have to connect it on two different PCs and write a client server program to sync them. It's a bummer since i think just 'knowing' the different kinects connected without accessing the camera for all of them at the same time would have solved the problem.

elliotwoods, 2015-11-06:
Maybe with some commands you can enable/disable devices (like you would in device manager).
You could identify which is which by its connection addresss

Sorry this can't be done right now

closed 2015-11-06 by elliotwoods