elliotwoods/ofxKinectForWindows2 #41: Build fails due to an unresolved external

Axagthoth, 2015-11-18:
Hi, @elliotwoods . I'm trying to use your addon on Windows 10 x64 and oF 0.9.0, but can't manage to build the example. The errors are:

LNK1120	1 unresolved externals	example	D:\of_v0.9.0_vs_release\addons\ofxKinectForWindows2\example\bin\example.exe

LNK2001	unresolved external symbol _GetDefaultKinectSensor@4	example	D:\of_v0.9.0_vs_release\addons\ofxKinectForWindows2\example\ofxKinectForWindows2Lib.lib(Device.obj)

I've tried commenting the lines in Device.cpp that make the call to GetDefaultKinectSensor(), but that only worsens the problem.

Honestly, I don't know what to do. The kinect SDK 1409 is installed, and I've confirmed the kinect works with the Kinect Studio and with Unity.

Thanks in advance!

cbtriplec, 2015-11-19:
Building a new project I get the same errors after following RBarraza's tutorial.  Cant seem to figure out the issue either.

elliotwoods, 2015-11-19:
OK i'll look into it
not sure what it is off hand and haven't got a working setup here right now
but will try and check this week.
did you make sure that the example project has a reference to the ofxKinectForWindows2Lib project?
sounds like ofxKinectForWindows2Lib project is missing references to the kinect library, which is strange if it's the only error which shows up

Axagthoth, 2015-11-20:
Thanks for answering so quickly!
I'm not exactly sure what you mean. When I open the example solution, three projects get loaded:
the example itself, ofxKinectForWindows2Lib, and openFrameworksLib. Doesn't this mean it has got the reference to the ofxKinectForWindows2Lib project? Or must the files generated by building ofxKinectForWindows2Lib be re-referenced somehow?

cbtriplec, 2015-11-21:
I have been adding the reference when building a new project, so I know its there, and still get there error.  I think the examples the reference has already been made so I think everything i set up properly on your end @Axagthoth

devtronslab, 2015-11-22:
I'm seeing this error, as well. Happens both in the example and in a new project. ofxKinectForWindows2Lib is definitely referenced in both, and the properties file is loaded, too. 

Windows 8.1
VS2015
OF0.9.0
Kinect2 works with Kinect Studio

devtronslab, 2015-11-22:
@elliotwoods Might be related to the props file. VS2015 is telling me that <Lib> element is an invalid child element for <ItemDefinitionGroup>

devtronslab, 2015-11-22:
Got it to work. Realized I was compiling for x86 target instead of x64. No changes needed to props file or anything else.

elliotwoods, 2015-11-22:
ah!!
good spot.
would you mind editing the Readme.md to reflect this issue?
(you can use the editor on GitHub to make changes, it'll automatically make
a PR for your changes)

cbtriplec, 2015-11-22:
Awsome!  Thanks for the help @devtronslab

Axagthoth, 2015-11-23:
@devtronslab @elliotwoods uuuhhh MAJOR facepalm for me. That was the issue! Sorry for not realizing earlier, and thanks to everybody. This is what makes oF great!

closed 2015-12-04 by elliotwoods