elliotwoods/ofxKinectForWindows2 #44: bodies[i].tracked set to true in init(), should be false

p4sI, 2015-11-30:
Hi,

just encountered that `bodies[i].tracked == true` after the init()-function in Body.cpp is executed. Even without a Kinect the state gets set to true. This happens after the line 47 is executed, where the bodies vector get resized from 0 to 6, see https://github.com/elliotwoods/ofxKinectForWindows2/blob/master/src/ofxKinectForWindows2/Source/Body.cpp#L47

I fixed it by adding 
```
for (int i = 0; i < BODY_COUNT; ++i)
{
	auto & body = bodies[i];
	body.clear();
}
```
after line 47.

PS: I'm using the 0.8.4 branch, someone should check if it happens in the master one as well

elliotwoods, 2015-12-03:
Hi @p4sI 
thanks for spotting this

the total issue here is that Data::Body does not have a default constructor (i.e. to set `tracked` to `false`, etc).
Let me fix this one.

p4sI, 2015-12-03:
Hi, 
okay this makes sense. I thought booleans are false by default and wondered why it gets set to true.

elliotwoods, 2015-12-04:
`bool`, `int`, `float`, etc have no default values in C/C++
they simply default to whatever was in memory there beforehand
therefore you should be careful always to initialise their values

elliotwoods, 2015-12-04:
thanks for the spot!

commit e2e6a064 (2015-12-04): Data::Body::Body() default constructor
Fixes #44
closed 2015-12-04 by elliotwoods