elliotwoods/ofxKinectForWindows2 #45: Error with pixels arrays type

karolsenami, 2016-01-04:
Hi, i'm having troubles with the exampleBodyIndexColor. 

It doesn't recognize the types of the arrays (declared as `auto&`). 
The declaration works if I put `<auto depthPix>` and not `<auto&>`.

But I still have the problem in the if statement. I tried using `->` instead of `.` but I still get the same error.

[image]

mattfelsen, 2016-01-04:
Hi @karolsenami, thanks for moving this discussion to GitHub :smile: I don't have access to a Windows machine at the moment but I'll do my best to help get you running.

You mentioned you're using VS2012 – are you using OF 0.8.4, 0.9.0, or something else? If you're on 0.8.4, make sure you checkout the `0.8.4` branch of the addon. This `exampleBodyIndexColor` project doesn't exist in that branch (it was added to the repo later), but if should hopefully work if you copy the code into the `example` project. Otherwise you can try upgrading to OF 0.9.0 and VS2015.

The variables on lines 45-47 are references, not pointers, so `.` is the correct syntax rather than `->`. I forget if the `auto` keyword is available in OF 0.8.4 so you can try replacing them with explicit variable types, but I am guessing the `master` branch vs `0.8.4` branch may be your issue. Can you comment on that and we'll go from there?

karolsenami, 2016-01-04:
Thanks for your quick answer. I've noticed that there was a `0.8.4` branch but I was getting a 404 error while trying to get on it so I thought that the master was `0.8.4`, my bad !

I tried copying the code in the `0.8.4 example` but I get the same errors. 

I then tried it on my OF 0.9.0, it works perfectly. I guess you should just specify somewhere that this example only runs on OF 0.9.0  :relaxed:

Thanks for your help.

closed 2016-01-04 by karolsenami

elliotwoods, 2016-01-04:
Did you try using
const auto &
It may be that VS2015 is adding the const automatically?