elliotwoods/ofxKinectForWindows2 #46: Two Body.cpp caused link problem

DeqingSun, 2016-01-12:
Two or more files with the name of Body.cpp will produce outputs to the same location. This can lead to an incorrect build result.  The files involved are ..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Data\Body.cpp, ..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Source\Body.cpp.

And looks like it is causing unresolved externals problem

elliotwoods, 2016-01-15:
aha
i guess the fix might somehow be missing from the project settings?
did you follow these settings?
http://rbarraza.com/adding-ofxkinectforwindows2-to-a-new-or-existing-of-project/

it sounds like you've added the ofxKinectForWindows2 to your project using the project generator (which doesn't have all the project settings ready for you).

Otherwise you need to add this setting:
```
https://github.com/elliotwoods/ofxKinectForWindows2/blob/master/ofxKinectForWindows2Lib/ofxKinectForWindows2Lib.vcxproj#L85
```
it should be somewhere in your Project Settings (Output Files section) if you go into project settings in Visual Studio

Elliot

DeqingSun, 2016-01-15:
Thanks
I followed the instruction and it worked well!
However I'm using VisualStudio 2015. I can not find "Add Existing Property Sheet". I had to open "ofxKinectForWindows2.props" in texteditor and add paths manually.

elliotwoods, 2016-01-15:
can you take a screenshot of your solution?

you need to do 'Add existing property sheet' in 'Property Manager' (it's
like Solution Explorer but different. i think there's a search box at the
top right of Visual Studio and you can just type 'Property Manger' there to
get to it)

but is sounds like your setup is still a bit off (maybe you don't have the
ofxKinectForWindows2Lib inside of your solution yet?)
did you do the 'Add reference' step?

elliot

elliotwoods, 2016-05-04:
Hi @DeqingSun 
sounds like your problem is solved
ok to close this issue?

closed 2016-05-05 by DeqingSun