elliotwoods/ofxKinectForWindows2 #47: UINT64 trackingId; instead of int trackingId in body.h?

shavmark, 2016-01-18:
I am tying Kinect face in with ofxKinectForWindows2 for one of my own projects and I think I found that the trackingID in ofxKinectForWindows2::Data::Body should be UINT64 and not int as it is now.  This is because in void Body::update() { ... UINT64 trackingId = -1; ... 	body.trackingId = trackingId; and as a result trackingId was not tying to the face tracking ID for me when i accessed it via Body later on

This is a small thing and I can change the code in my local version but I wanted to let you know and to thank you very much for sharing this great code

elliotwoods, 2016-05-04:
thanks @shavmark 
can you check the most recent version
https://github.com/elliotwoods/ofxKinectForWindows2/blob/801caa732e6b53caa9d0d9244c0de684b37fc278/src/ofxKinectForWindows2/Data/Body.h

now we're using `uint64_t`

(we still default to `-1` in Body.cpp)

shavmark, 2016-05-04:
yes, that type change should do it, thanks!

closed 2016-05-04 by shavmark