elliotwoods/ofxKinectForWindows2 #48: ofxKinectForWindows2 and ofxFlowTools

rootux, 2016-02-18:
I've tried combining ofxKinectForWindows2 and [ofxFlowTools](https://github.com/moostrik/ofxFlowTools/) to read data from the kinect camera.
https://github.com/rootux/visionquest
I'm calling the kinect like so:
`kinect.getColorSource()->draw(0, 0, cameraFbo.getWidth(), cameraFbo.getHeight());`
I've tried the IR and the depth - all seem to be drawing the camera OK **without** going through the filters (see attached [Image](http://i.imgur.com/NU0PK9Y.jpg) )

What am I missing?

elliotwoods, 2016-02-19:
Hey @rootux . It looks like the ofxKinectForWindows side is working fine
but I haven't used ofxFlowTools so I don't know why that bit isn't working

did you try switching this line to use a webcam instead?
https://github.com/rootux/visionquest/blob/master/src/ofApp.cpp#L182

(just to check that it's definitely a quirk related the Kinect image)

elliotwoods, 2016-05-04:
hey @rootux 
closing here as it seems this likely isn't an issue with the current version of ofxKinectForWindows2

closed 2016-05-04 by elliotwoods