elliotwoods/ofxKinectForWindows2 #49: ofFloatPixels getDepthToWorldMap() is filled with negative values

benkuper, 2016-02-24:
Hello,
i'm trying to get a worldMap from the kinect, but i can't find any simple function like getWorldPos(ofPoint depthCoord) to convert a position in pixel from the depth image to a world coordinate.

I saw the >kinect.getDepthSource()->getDepthToWorldMap() function but the resulting ofFloatPixels object is filled with -431602080.000000

do you know why is that, or am i not using it right ?

thanks

benkuper, 2016-02-25:
i managed to get something working with the ICoordinateMapper finally.

elliotwoods, 2016-02-25:
hey!
maybe you're using an old version of the addon?
the function you're looking for in the current version is `getWorldInDepthFrame`
https://github.com/elliotwoods/ofxKinectForWindows2/blob/master/src/ofxKinectForWindows2/Source/Depth.cpp#L146

elliotwoods, 2016-02-25:
and sorry for not replying earlier (i thought i had)

closed 2016-05-04 by elliotwoods