elliotwoods/ofxKinectForWindows2 #50: Unable to run the bodyIndexShader example in VisualStudio2015

ConnectedCat, 2016-04-04:
Trying to run the code from exampleBodyIndexShader in VS2015, get the following error - see attached. I tried running the same exact code in a brand new project setup - there is no error in this case.
[image]

mattfelsen, 2016-04-04:
Hey @ConnectedCat, this is a funny Visual Studio issue that has bit me before too:
https://forum.openframeworks.cc/t/unable-to-start-program-ofxoculusdk2/18429/8

Try right-clicking on `example-bodyIndex` in the sidebar (the Solution Explorer) and choose "Set as StartUp project" – you should the bolded project change from `openframeworksLib` (according to your screenshot) to the example project. Hopefully that gets the app starting – no guarantees the example itself is in working order though :stuck_out_tongue_closed_eyes:

elliotwoods, 2016-04-05:
> the example itself is in working order though [image:

+1 to what matt said.

Please try and let us know if it still has a problem

ConnectedCat, 2016-04-14:
Yep, works now. Thank you much!

closed 2016-04-15 by elliotwoods