elliotwoods/ofxKinectForWindows2 #51: Errors compiling/linking example

ojack, 2016-04-11:
I am having trouble running the example and getting the following errors:
[image]

Using: 
Visual Studio 2015
OpenFrameworks 9.0.3
Kinect V2 that works with SDK

(I am new to developing with OpenFrameworks on Windows---apologies if this is an obvious mistake) Thanks!

mattfelsen, 2016-04-11:
Hi @ojack. At a (super quick) glance it looks similar to something I ran into previously. Can you try moving the OF folder as mentioned in this comment and see if that's it?
https://github.com/elliotwoods/ofxKinectForWindows2/issues/36#issuecomment-139840620

elliotwoods, 2016-04-11:
Hi @ojack . I think @mattfelsen has spotted your issue (matt's on fire!!)

One of the weird VS / windows issues is that it doesn't support very long path names
so if you're in like c:\users\awesomemomma\Documents\My Projects\openFrameworks\addons\ofxBlah
then it can cause issues

I generally use c:\openFrameworks\addons\ofxBlah

Maybe there's a Visual Studio setting somwhere we can change for oF projects to handle this
but maybe we just have to wait for VS2017 and keep our fingers crossed they fix it...

elliotwoods, 2016-04-11:
let me know how it goes!

ojack, 2016-04-11:
thanks for the quick replies @mattfelsen @elliotwoods 

I tried moving the folder but am still getting a link error :/ 
[image]

ojack, 2016-04-11:
it seems like maybe it is looking in the wrong place for ofxKinectforWindows2Lib? the example/bin folder is empty... 

(But ofxKinectforWindows2Lib shows up in the Resource View and is added as a reference.) 
any ideas?

thanks again :)

ojack, 2016-04-12:
It works when I comment out OfGetCurrentWindow() in device.cpp of ofxKinectforWindows2Lib

icq4ever, 2016-04-27:
I have a same issue.. and still I canmot solve it.

Axagthoth, 2016-04-28:
I'm having the same issue too. I followed @ojack 's advice, and with line 182 of Device.cpp looking like this:

`auto mainWindow = std::shared_ptr<ofAppGLFWWindow>(/*ofGetCurrentWindow()*/);`

...it seems to work for me. Directory tree didn't cut it, as I already had the folder at C:\openFrameworks. I do get a couple  of these though:

`[warning] ofGLRenderer: draw(): texture is not allocated`

fralken, 2016-05-02:
I noticed that `ofGetCurrentWindow()` is not available in OF 0.9.3. It has been added in a recent commit on github.

For me the easiest solution was to manually add this method in `ofAppRunner` (.h and .c)

Somewhere in `ofAppRunner.h`:

    std::shared_ptr<ofAppBaseWindow> ofGetCurrentWindow();

Somewhere in `ofAppRunner.cpp`:

    std::shared_ptr<ofAppBaseWindow> ofGetCurrentWindow() {
    	return mainLoop()->getCurrentWindow();
    }

An alternative is to use the master on Github of OF instead of the 0.9.3 package.

Axagthoth, 2016-05-03:
Hey @fralken thanks a lot for the heads up! The in Device.cpp type still needs to be changed to std::shared_ptr though, right?

fralken, 2016-05-03:
No, you don't need to change Device.cpp

elliotwoods, 2016-05-04:
I've added a version guard around the `ofGetCurrentWindow` check so it should be backwards compatible now with versions of oF which don't have that function
https://github.com/elliotwoods/ofxKinectForWindows2/blob/555e77dc0e8c2f23ec63d1345798b236cc642f79/src/ofxKinectForWindows2/Device.cpp#L186

The linker error shown in @ojack's screenshot:
[image]
shows that ofxKinectForWindows2Lib failed to build (because it couldnt' build Device.cpp) and therefore the main app couldn't link the addon lib

elliotwoods, 2016-05-04:
this should be fixed now. just pull latest. Device.cpp should build and the linker error should disappear

carloscastellanos, 2017-03-08:
Hey guys I am having a similar problem and I think part of the issue is that the downloads for openFrameworks on the openFrameworks site are out of date (at least those for Visual Studio and XCode).

For example these are the two errors I get when I try to build:
[image]

This is happening because the ofGraphics.h and .cpp files that I downloaded from the ofx site do not contain the ofRotateDeg() methods while the same files on the Github repo clearly do. As it stands now none of the ofxKinectForWindows2 examples will compile due to this error.  I am going to submit this issue to the main ofx GitHub repo as well but I think a workaround would be to download ofx directly from the GitHub repos and just add the VS project/solution files.

mattfelsen, 2017-03-08:
Hey @carloscastellanos. Are you using a 0.9.x release version of OF, or the development version from GitHub? If you're using a release, try checking out the `0.9.0` tag of this repo.

The `master` branch of this repo is set to work with the dev version of OF (`ofRotate` has been replaced with separate versions for `ofRotateDeg` (degrees) or `ofRotateRad` (radians) so I think that's the issue you're running into

carloscastellanos, 2017-03-08:
O damn, how did I miss that? It compiles fine now. Sorry!

mattfelsen, 2017-03-08:
No worries! Glad it was an easy fix :)

closed 2017-03-08 by elliotwoods