elliotwoods/ofxKinectForWindows2 #52: 'Cannot open include file: 'Kinect.h': No such file or directory'

ConnectedCat, 2016-04-15:
I'm not sure if this is the issue with the ofxKinectForWindows2 library or a more general VS2015 quirk but I get this error under the following circumstances:
1) Create a new project in openFrameworks projectGenerator - include ofxKinectForWindows2 library
2) Follow instructions from https://github.com/elliotwoods/ofxKinectForWindows2#usage
3) Compile an empty project

Sometimes the order in my Solutions Explorer comes out as follows:
 - ofxKinectForWindows2
 - openframeworksLib
 - projectName

in which case I get the following errors:
[image]

But sometimes the order comes up like this:
 - projectName
 - ofxKinectForWindows2
 - openframeworksLib

and then everything complies fine. The Build Order is the same in both cases:
 - openframeworksLib
 - ofxKinectForWindows2
 - projectName

and the Dependencies are the same. What does the order in the Solutions Explorer mean and how do I fix it?

elliotwoods, 2016-04-15:
Please make sure:

1. You have the .props file attached to your videoForTest project
2. You have Kinect 2.0 SDK installed
3. You have rebooted since installing the SDK (this is necessary to flush the paths)

ConnectedCat, 2016-04-17:
Thanks for looking into this. All of those three check out fine. I've had the SDK installed for a while and rebooted multiple times since. Also, the compile problems occur on simultaneously open projects - some compile, some do not. The only visible difference I could find is the order of components in the solution.

ojack, 2016-04-23:
@ConnectedCat Are you possibly creating the project within the "addons" folder, rather than in the apps folder? I get similar errors as well as the order change when I do that.

ConnectedCat, 2016-04-26:
@ojack No, both projects are in Documents\openFrameworks\apps\myApps

elliotwoods, 2016-05-04:
hey @ojack @ConnectedCat 
i'm curious to know what this error might be
did either of you make any progress identifying the issue

interesting that you can't make projects inside addons folder (this works fine for me here)

elliot

grrrwaaa, 2016-05-06:
We had the same issue, also with a project created via the VS/ofx project generator (missing Kinect.h) but it was resolved by manually adding the ofxKinectForWindows2.props in the VS property manager pane.

However we still have linker errors:

1>Body.obj : error LNK2019: unresolved external symbol "public: __cdecl ofxKinectForWindows2::Data::Body::Body(void)" (??0Body@Data@ofxKinectForWindows2@@QEAA@XZ) referenced in function "public: void __cdecl std::allocator<class ofxKinectForWindows2::Data::Body>::construct<class ofxKinectForWindows2::Data::Body>(class ofxKinectForWindows2::Data::Body *)" (??$construct@VBody@Data@ofxKinectForWindows2@@$$V@?$allocator@VBody@Data@ofxKinectForWindows2@@@std@@QEAAXPEAVBody@Data@ofxKinectForWindows2@@@Z)
1>Body.obj : error LNK2019: unresolved external symbol "public: void __cdecl ofxKinectForWindows2::Data::Body::drawWorld(void)const " (?drawWorld@Body@Data@ofxKinectForWindows2@@QEBAXXZ) referenced in function "public: void __cdecl ofxKinectForWindows2::Source::Body::drawWorld(void)" (?drawWorld@Body@Source@ofxKinectForWindows2@@QEAAXXZ)
1>Body.obj : error LNK2019: unresolved external symbol "public: void __cdecl ofxKinectForWindows2::Data::Body::clear(void)" (?clear@Body@Data@ofxKinectForWindows2@@QEAAXXZ) referenced in function "public: virtual void __cdecl ofxKinectForWindows2::Source::Body::update(void)" (?update@Body@Source@ofxKinectForWindows2@@UEAAXXZ)
1>Body.obj : error LNK2019: unresolved external symbol "public: static class std::vector<struct std::pair<enum _JointType,enum _JointType>,class std::allocator<struct std::pair<enum _JointType,enum _JointType> > > const & __cdecl ofxKinectForWindows2::Data::Body::getBonesAtlas(void)" (?getBonesAtlas@Body@Data@ofxKinectForWindows2@@SAAEBV?$vector@U?$pair@W4_JointType@@W41@@std@@V?$allocator@U?$pair@W4_JointType@@W41@@std@@@2@@std@@XZ) referenced in function "public: void __cdecl ofxKinectForWindows2::Source::Body::drawProjected(int,int,int,int,enum ofxKinectForWindows2::ProjectionCoordinates)" (?drawProjected@Body@Source@ofxKinectForWindows2@@QEAAXHHHHW4ProjectionCoordinates@3@@Z)
1>Device.obj : error LNK2019: unresolved external symbol GetDefaultKinectSensor referenced in function "public: void __cdecl ofxKinectForWindows2::Device::open(void)" (?open@Device@ofxKinectForWindows2@@QEAAXXZ)
1>bin\shpox_debug.exe : fatal error LNK1120: 5 unresolved externals

Building debug, x64. 

I notice that we are also seeing the "two body.cpp" warning, which might partially explain this. But the (closed) issue documenting it points to urls that don't seem valid. How do we solve it?

1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppBuild.targets(936,5): warning MSB8027: Two or more files with the name of Body.cpp will produce outputs to the same location. This can lead to an incorrect build result.  The files involved are ..\..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Data\Body.cpp, ..\..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Source\Body.cpp.
1>  main.cpp
1>  ofApp.cpp

elliotwoods, 2016-05-06:
> __cdecl std::allocator::construct(class ofxKinectForWindows2::Data::Body
> class std::vector,class std::allocator > > const & __cdecl

did you add a reference to the ofxKinectForWindows2Lib project to your app
project?

i.e.
* Right click on your project
* Select 'Add Reference'
* Make sure that the ofxKinectForWindows2Lib project is selected (as well
as openFrameworksLib)

grrrwaaa, 2016-05-06:
We tried to, but there was nothing else in the Add Reference pane other than openFrameworksLib

We worked around it by copying one of the examples into our apps folder, but we needed to bring the ofxKinectForWindows2 props and lib folders too.

(Also needed to manually copy the non-debug lib into our project's bin to build Release)

ojack, 2016-05-06:
@grrrwaaa you also need to add the project ofxKinectForWindows2 to the solution, and then it should be available as a reference

ConnectedCat, 2016-05-08:
@elliotwoods I'm trying to solve this by exclusion, but can't find any repeatable causes of error. Primary suspect right now is ofProjectGenerator. Tried the following:
1) Made a new project, added project, property sheet and reference - complied fine.
2) Quit ofProjectGenerator, reopened, repeated step 1 - compiled fine.
3) Made a new project without quitting ofProjectGenerator, added project, property sheet and reference - 'Kinect.h' error on compile
here is where it gets weird:
4) Quit ofProjectGenerator, reopened, repeated step 1 - 'Kinect.h' error on compile
5) Restarted machine, reopened ofProjectGenerator, repeated step 1 - 'Kinect.h' error on compile

Not even sure where to go from here. The only repeatable correlation is between the compile errors and the order of dependencies, as I described in the opening comment.

wezside, 2016-05-18:
I've been struggling with the exact same issue. A new project created using the Project Generator would throw the Kinect.h error. Here is how I solved it using VS2015 - hopefully it helps someone else.

1. Install Kinect SDK (possible restart required)
2. Download add-on 
3. Generate a new project using Project Generator and add ofxKinectForWindows2 addon
4. Open project
5. In solution explorer - right click on Solution and choose Add -> Existing Project
6. Navigate to the **addon\ofxKinectForWindows2\ofxKinectForWindows2Lib\** and choose the .vcxproj
7. In solution explorer - right click on your project Add -> Reference...
8. Check/enable ofxKinectForWindows2Lib

This next bit is where I stumbled just because I don't know VS2015.You must add the add-on property file to your solution:

9. View -> Other Windows -> Property Manager
10, Click on the icon with a green plus in the Property Manager (Add existing prop sheet)
11. Browse to **addons\ofxKinectForWindows2\ofxKinectForWindows2.props**

Compile and everything should work.

wezside, 2016-05-18:
So while the above steps work it will reset any existing addons you chose during project creation. I wanted ofxOsc and that was not found after doing the above. To fix this instead create your project, load the props file first, update your project using the generator with all add-ons selected, then follow the steps of adding the project as reference.

ConnectedCat, 2016-05-25:
@wezside Just tried what you suggested - didn't work.
It seems that your sequence is exactly the same as described here: https://github.com/elliotwoods/ofxKinectForWindows2#usage
except to the order or steps 2 and 3.
It really seems to me that that during the compilation VS2015 is trying to access Kinect.h in the app before the ofxKinectForWindows2, which contains a reference to it, is built. And I don't know how to change the sequence.

wezside, 2016-05-25:
@ConnectedCat  What happens if you load the property sheet last?

ConnectedCat, 2016-05-25:
@wezside fails the same way regardless of the order.
I just tried making a new project without the openFrameworks projectGenerator - directly in VS2015. Same scenario - fails with 
Error	C1083	Cannot open include file: 'Kinect.h': No such file or directory (compiling source file ..\..\..\..\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Data\Body.cpp) etc....

ConnectedCat, 2016-05-25:
It seems that I have found a reliable workaround:
When creating a new project, whether in VS2015 or in oF projectGenerator - DO NOT include the ofxKinectForWindows2 library. Make a plain project and then follow the instructions from https://github.com/elliotwoods/ofxKinectForWindows2#usage, i.e.:
 - Add the ofxKinectForWindows2Lib.vcxproj to your solution
 - In Property Manager (open it from View -> Other Windows -> Property Manager), right click on your project to select Add Existing Property Sheet... and select the ofxKinectForWindows2.props file
 - Right click on your project (e.g. 'mySketch') and select 'Add Reference...', and add a reference to ofxKinectForWindows2Lib.

I dunno why it works but it works. I'd like to know why though, please let me know if you understand the mechanism here.

fralken, 2016-05-26:
There must be some problem when you add the ofxKinectForWindows2.props property sheet, because this is where "Additional Include Directories" are defined.

In Property Manager, expand the tree of your project and double click on ofxKinectForWindows2. 
In "Common Properties -> C/C++ -> All Options" you should see that "Additional Include Directories" references the Include dirs of Kinect SDK and this addon.

Also, in the Property Page of your project, that same entry should contain also %(AdditionalIncludeDirectories), for all configurations.

tdubnov, 2016-11-09:
Was anyone able to resolve this issue?
I'm having the same problem. The examples manage to build but when I include it in my own project I get "cannot open source file Kinect.h"

Any help is greatly appreciated!

elliotwoods, 2016-11-09:
@fralken is right here. any error relating to Kinect.h not being found can come from either:

1. Kinect.h is not in your include paths (then you need to add the `props` file as mentioned in the readme)
2. Kinect.h is not on your system (then you need to install Kinect __SDK__ and then restart)

@wezside - you can add ofxKinectForWindows2 to an existing project using the instructions in the readme, you don't need to start with a fresh project (unless you think you might have broken some of your project settings somewhere)

@ConnectedCat - again looks like you don't have the `props` file setup properly. I'm not sure what you're doing wrong, but perhaps you've added the `props` file to the wrong project? (e.g. not your 'myApp' ). Please send a screenshot of your project in the prop

@tdubnov - check the Readme for the instructions. it seems you've missed something

elliotwoods, 2016-11-09:
I've added a troubleshooting section to the Readme.md to help people along

closed 2016-11-09 by elliotwoods

AvinandanBandyopadhyay, 2018-07-17:
Sometimes the solution can be as simple as changing the platform from x86 to x64 in VisualStudio or vice-versa.

renhaofan, 2018-08-05:
I'm working on the environment:
OS: Ubuntu16.04
IDE: Clion2018.1
I met the same problem, but the solution is for windows. please help me !

elliotwoods, 2018-08-07:
@renhaofan
This add-on is Windows only, as it relies on the official Microsoft SDK which is Windows only

commit b9635dc9 (2016-11-09): readme updates to troubleshoot. closes #52