elliotwoods/ofxKinectForWindows2 #53: some problems with the project setup

bluelemonade, 2016-04-20:
hello elliot,

nice to see your great work, you did last year a small v4 patch for me, you remember? Now I try to use the Kinect2 for simple user tracking to controll a few spots.

I made a new project inside "E:\of_v0.9.3_vs_release\apps\myApps" with the project generator.
after this I put the ofxKinectForWindows2.props inside this folder and add it to my project in vs2015.

know I added the project file from "E:\of_v0.9.3_vs_release\addons\ofxKinectForWindows2\ofxKinectForWindows2Lib" to my solution.
(should I copy it to the folder or use the one inside the addon folder?)

my src files are totally empty. and i tried to compile this empty project.

[image]

where's this error coming from:
Cannot open include file: 'ofxSingleton.h': No such file or directory (compiling source file ..\..\..\..\addons\ofxCvGui\src\ofxCvGui\Element.cpp)	openframeworksLib	e:\of_v0.9.3_vs_release\addons\ofxcvgui\src\ofxcvgui\Utils\Utils.h

how can I fix this issue?

regards dietmar

elliotwoods, 2016-05-04:
it seems that ofxCvGui is in your project somehow
did you start by copying the example with gui?
perhaps you want to check that none of the ofxCvGui source files are in your project anywhere

just doing a 'clean solution' 'build solution' doesn't fix it?
check also that ofApp.h doesn't include ofxCvGui by accident

or search 'ofxCvGui' in the solution to see if it's being included by accident somewhere

bluelemonade, 2016-05-06:
ok, I installed 0.9.3 and copied only the new downloaded ofxkInectWindows2 in the addon folder.

created a clean project with the generator, added the props to my example. 
now I got some linker errors:

[image]

Using the example direct from the addon folder generates a roboycopy error??? 

"robocopy "E:\of_v0.9.3_vs_release\libs\openFrameworksCompiled\project\vs\\..\..\..\..\/export/vs/x64/" "E:\of_v0.9.3_vs_release\addons\ofxKinectForWindows2\ofxKinectForWindows2Lib\bin/" "*.dll" /njs /njh /np /fp /bytes
if errorlevel 1 exit 0 else exit %errorlevel%

hmmm...

bluelemonade, 2016-05-09:
hi,

I fixed this issue, wrong robocopy version! Now I can compile the compile example and the exampleBodyIndexColor example. yeah!

the next example are stopping with this message box:

Unable to start program 'E:\of_v0.9.0_vs_release\libs\openFrameworksCompiled\project\vs\..\..\lib\vs\x64\openframeworksLib_debug.lib'.

or: 
E:\of_v0.9.0_vs_release\libs\openFrameworksCompiled\project\vs\..\..\lib\vs\x64\openframeworksLib_debug.lib is not a valid Win32 application.

dietmar

elliotwoods, 2016-05-10:
glad that you made a lot of progress!
sorry for not being there to help at the time

it sounds like:

1. You might have openFrameworks set as your startup project (right click on your app project and select 'set as startup'
2. You're compiling as Win32 rather than x64 (but i think 1 is more likely)

bluelemonade, 2016-05-10:
ok, I cleaned the project and recompiled, now the exampleBodyIndexShader works

the exampleDepthToCameraTable eample shows a "load failed" in the solution window. when reloading I get this:

---------------------------
Microsoft Visual Studio
---------------------------
Unable to read the project file "exampleDepthToCameraTable.vcxproj".

E:\of_v0.9.0_vs_release\addons\ofxCvGui\ofxCvGuiLib\ofxCvGui.props(5,5): The imported project "E:\of_v0.9.0_vs_release\addons\ofxAudioDecoder\ofxAudioDecoder.props" was not found. Confirm that the path in the <Import> declaration is correct, and that the file exists on disk.
---------------------------
OK   
---------------------------

regards.

bluelemonade, 2016-05-10:
ok, i put a lot of amazing things from your github account into the addons...  now I only get these errors:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2228	left of '.glVersionMajor' must have class/struct/union	ofxCvGuiLib	e:\of_v0.9.0_vs_release\addons\ofxcvgui\src\ofxcvgui\panels\texture.cpp	41	
Error	C3536	'mainWindow': cannot be used before it is initialized	ofxCvGuiLib	e:\of_v0.9.0_vs_release\addons\ofxcvgui\src\ofxcvgui\panels\texture.cpp	39	
Error	C3861	'ofGetCurrentWindow': identifier not found	ofxCvGuiLib	e:\of_v0.9.0_vs_release\addons\ofxcvgui\src\ofxcvgui\panels\texture.cpp	38	
Error	LNK1104	cannot open file 'E:\of_v0.9.0_vs_release\addons\ofxCvGui\ofxCvGuiLib\bin\Debug\ofxCvGuiLibd.lib'	example	E:\of_v0.9.0_vs_release\addons\ofxKinectForWindows2\exampleWithGui\LINK	1	
Error	C2227	left of '->getSettings' must point to class/struct/union/generic type	ofxCvGuiLib	e:\of_v0.9.0_vs_release\addons\ofxcvgui\src\ofxcvgui\panels\texture.cpp	41

elliotwoods, 2016-05-11:
Ah! I think it's because you're using the latest branch of ofxCvGui. Which
needs the most recent version of openFrameworks (from GitHub) sorry!!

bluelemonade, 2016-05-11:
hi,

can I use an older version with 0.9.0

regards

16:56 Uhr:

elliotwoods, 2016-05-11:
if you want you can ignore the exampleWithGui, no?
the API might have changed, but maybe try to get a commit of ofxCvGui before January og this year
but since it relies on other addons also, it could get complicated quickly, so suggest leave the gui example alone for the time being :)

bluelemonade, 2016-05-11:
your right, I try to check the other examples.

one question: when I call body.joints[JointType_Head].getOrientation(); I
should get back a ofQuaternion. But I get all 4 values with 0.

17:31 Uhr:

elliotwoods, 2016-05-11:
could you try putting a breakpoint around here:
https://github.com/elliotwoods/ofxKinectForWindows2/blob/master/src/ofxKinectForWindows2/Source/Body.cpp#L127

and see if the SDK is giving us empty orientations? (i.e. the addon has a problem later on)

i haven't used orientations so far

bluelemonade, 2016-05-12:
hi,

strange, only the head is without an orientation. 

regards

[image]

mattfelsen, 2016-05-12:
@bluelemonade This is reminding me of something a friend ran into on a project...I don't remember the details, but I think it was somewhere the the head join rotation is always zero (like you're seeing), and you instead have to get it either via a neighboring joint (the neck?) or use the face tracking API. Maybe you can try looking into those?

bluelemonade, 2016-05-13:
ok, face recognition isn't already implemented in the ofxKFW2 plugin?

mattfelsen, 2016-05-13:
No, but there are a couple forks where others have added support. You could check the [network graph](https://github.com/elliotwoods/ofxKinectForWindows2/network) to take a look. My buddy was using [this](https://github.com/local-projects/ofxKinectForWindows2/commit/8446d6658dfa25a796b9de0d34db68131da3c4840), which he wrote. 

I would try the neck joint orientation first though – a much simpler thing to check and would be much easier than worrying about the face tracking.

bluelemonade, 2016-05-16:
thanks, I try the neck or the orientation of the shoulders.