elliotwoods/ofxKinectForWindows2 #54: can not add ofxKinectForWindows2.props

fdiba, 2016-04-23:
Hello,

I get few compile errors while trying to make work the first example. I think it is because I didn't add the "ofxKinectForWindows2.props" file to my project but I don't know how to do it.

Can you help me ?

I can't find the label "Add Existing Property Sheet..." in the "Property Manager" as explained in the readme file.

[image]

elliotwoods, 2016-04-23:
It seems to be an openFrameworks version mismatch
I think I'm using a function there which isn't in 0.9.0
It's not harmful to remove that part. Just search for where the error is
coming up and remove the offending lines
They'll be related to an if statement
Just make it true instead of checking this missing function

fdiba, 2016-04-23:
Thank you, it works!

This is the line that causes the problem in Device.cpp:

`auto mainWindow = std::static_pointer_cast<ofAppGLFWWindow>(ofGetCurrentWindow()); `

closed 2016-04-23 by fdiba