elliotwoods/ofxKinectForWindows2 #55: Multi sources

fralken, 2016-05-02:
Hello Elliot,
I'm implementing a green screen effect (remove the background) with custom background with kinect v2. At first I used WPF with C# (and F# also), but then I diceded to have a try with OpenFrameworks since it gives a lot of flexibility.
Your fantastic addon is exacltly what I need, and it has a green screen effect example too!

I though noticed that sometimes color frame and depth frame are not in sync.
It means that these frames are not available at the same time. In my opinion the `isFrameNew` flag of `Device` should be updated in `and` rather than `or`: `this->isFrameNewFlag &= source->isFrameNew();
` instead of `this->isFrameNewFlag |= source->isFrameNew();`.

No problem, I can check `isFrameNew` flag for each source explicitly with `&&`.

Then I added a call to the handy utility `kinect.getBodySource()->drawProjected` in the `draw()` method to draw the body skeleton, very nice to have a single call that does the job.
But again I noticed that sometimes the body is not in sync with the other frames.

I thought that maybe in this case should be better to use the MultiSourceFrameReader, which guarantees that all the desired sources are in sink.

But your abstraction keeps the Source and the Frame in the same object, so it is not easy to add a MultiSource implementation.
A solution could be to split Sources and Frames in separate objects (but some refactoring is required), in such a way that a Source can provide the corresponding Frame, while a MultiSource can provide Frames which a Reader as been opened for.

What do you think about it?
Thanks
francesco

elliotwoods, 2016-05-04:
agreed it's a good idea to implement multi-sources. probably even as the default usage pattern

i think in your case right now, it makes more sense to check the source isFrameNew flags yourself.
(i decided to use OR for isFrameNew on the device class)

i heard that in general, the kinect v2 doesn't guarantee synchronisation between frames
but in your experience it does keep sync with multisourceframereader?

elliotwoods, 2016-05-04:
i probably would keep the Source classes as they are, and implement the MultiSourceFrameReader in the Device class, which would populate the data in the Source classes (therefore keeping the API the same)

fralken, 2016-05-04:
Thanks @elliotwoods for your reply.
I noticed that with MultiSourceFrame, sources appear to be in sync, while when I acquire frames separately, sometimes they are slightly not in sync. Not a big deal, but it would be nice to expose the MultiSource functionality.

After some thoughts I agree that it's better to have the global isFrameNew in OR and optionally check isFrameNew  for the sources explicitly.

I also agree that, rather than splitting Sources and Frames, separating the Reader from the Source can have a minor impact on the codebase. 
I made an implementation where I split the `update()` method so that the `acquireFrame` and the update of the frame are separated, like so:

```
// pseudo code
// sources have these update methods:

update(FrameType * frame) {
    // here it is the logic for updating the data from the frame
   ...
}

update() {
    // pseudo code
   frame = acquire frame from the source reader
   update(frame); // then call the previous update
}

update(IMultiSourceFrame * frame) {
    // the Device passed a multi source frame
   frame = extract the frame from the multi source frame
   update(frame); // then call the previus update
}

```

Then the `Device` can update the sources either by calling their `update()` method, or by first acquiring the MultiSourceFrame and then passing it to the sources.

This doesn't change the API in any way for the user, apart that now the user can call a single init like so:

```
	kinect.initMultiSource({
		FrameSourceTypes_Depth,
		FrameSourceTypes_Color,
		FrameSourceTypes_Infrared,
		FrameSourceTypes_Body,
		FrameSourceTypes_BodyIndex });

```
Everything else stays the same.
You can also mix so that some sources use the MultiSourceReader and others use their own (I don't know a real use case for this though).

I still have to do some testing and then I'm going to publish it. I'll show you if you have time.

closed 2016-11-09 by fralken