elliotwoods/ofxKinectForWindows2 #62: exampleDepthToCameraTable : load failed

fdiba, 2016-06-10:
Hello,

I tried to load few examples : some are working other don't.

Like exampleDepthToCameraTable. It fails to load. When I try to reload it it says :

F:\of_v0.9.3_vs_release\addons\ofxKinectForWindows2\exampleDepthToCameraTable\exampleDepthToCameraTable.vcxproj : error  : Unable to read the project file "exampleDepthToCameraTable.vcxproj".

Do you know how I can make it work ?

elliotwoods, 2016-06-10:
That example relies on ofxCvGui
ofxCvGui relies on some other addons also

Let's leave this issue open until ofxCvGui is removed from the example

For the time being perhaps try to recreate the example without ofxCvGui by
looking at important bits of ofApp.cpp directly

fdiba, 2016-06-11:
Ok thank you for the explanation : I tried to load ofxCvGui but I didn't see it needed also another addon.

closed 2021-04-03 by elliotwoods