elliotwoods/ofxKinectForWindows2 #63: How to assign a body

Technolife2, 2016-08-26:
First of all, sorry to post here, but I didn't find anything on internet that could help me, so...

I'd like to track bodies and assign them to an ableton live track by the order of appearance. 
Like : 
First person detected -> AL Track 1
Second person detected -> AL Track 2

To achieve this, I've tried to use the numBodiesTracked : 

```
numBodiesTracked = 0;
auto bodies = kinect.getBodySource()->getBodies();
for (auto body : bodies) {
	if (body.tracked) {
		numBodiesTracked++;
		ofxAbletonLiveTrack *track = live.getTrack(numBodiesTracked - 1); //Track 1 is index 0
         }
}

```
But it seems that if the bodyID of the new person/body is lower than the previous one, the new body will be pushed before the old, resulting that the second body will have the Ableton track 1 instead of 0. 
Like : 
```
1st person -> bodyId 3 -> Ableton Track 0
then 
1st person -> bodyId 3 -> Ableton Track 0
2nd person -> bodyId 4 -> Ableton Track 1

```
and : 
```
1st person -> bodyId 3 -> Ableton Track 0
then 
2nd person -> bodyId 2 -> Ableton Track 0
1st person -> bodyId 3 -> Ableton Track 1

```

Does anyone already faced this ? 

Thank !
T

elliotwoods, 2016-08-28:
hi @Technolife2 

It sounds like you want to sort the bodies by the order in which they appeared, and should use the `trackingID` member of `Data::Body`, e.g:

```cpp
void myApp::update() {
    ...
    //make a map to store all the bodies in a sorted order
    map<int, ofxKinectForWindows2::Data::Body> bodiesSortedByAge;

    //go through the bodies, storing them using trackingId as key
    for (auto body : bodies) {
        if(body.tracked) {
        	//ok we've got a tracked body, now we need to 
        	bodiesSortedByAge[body.trackingId] = body;
        }
    }

    for(auto body : bodiesSortedByAge) {
    	//now we have the bodies sorted by age
    }
}
```

The issue you might encounter is that your tracks will still jump (this time, when the first person leaves, the second person will jump over to the first ableton track).

Generally the best way is to use pre-assigned slots (e.g. always have 6 ableton tracks and fade them up/down). Note that the vectory returned by `getBodies()` is already slot assigned. There'll always be 6 slots (i think that's right), so if you have the same number of ableton tracks as that, then you can always assign a body to a track, and not have it jump around as bodies appear and disappear.

Elliot

KyrillosL, 2016-09-17:
Hi Elliot, 

Thank a lot for this really fast response ! 
I was in vacation and had a little break up, I'll try this right now. 

Kyrillos

elliotwoods, 2016-09-17:
Sorry to hear about the break up. (Unless that's a type and you just had a
little break, not break up)
Shall I close this then?

closed 2017-03-14 by elliotwoods