elliotwoods/ofxKinectForWindows2 #65: coordinate mapping

rwebber, 2016-09-28:
I have managed to port your example using coordinatemapper to my project.. and its great.
I have a green screen effect being sent via spout.
However, the loops are slowing down the frame rate to ~20fps
I would like to be able to have a mapped color feed.. without a test for each pixel (in or out of body).
That lines up with the depth image.
Can this be done? I saw that MapColorFrameToCameraSpace is used in the depth.cpp

maybe I miss understand its use?

Sorry, to ask this here. I'm a newb, and searching hasn't found me much that is clear.

mattfelsen, 2016-09-28:
Hi @rwebber. Yes, the per-pixel operations on the CPU are indeed slow, but hopefully easy to understand what's going on. The faster way to do is in on the GPU with a shader.

OF includes an example for using a shader to reveal parts of an image with a mask. You could check that out and modify it to fit your needs. One thing to note is the mask you'd be using, the BodyIndex image, is actually inverted from a typical mask. Usually black pixels would be ones that you'd want to discard, and white ones you'd keep. In the BodyIndex image, white pixels are the background (and can be discarded) and anything else corresponds to a body.
https://github.com/openframeworks/openFrameworks/tree/master/examples/gl/alphaMaskingShaderExample

Another relevant example is in this repo – the `exampleBodyIndexShader` project – It shows how to pass the BodyIndex and Color images to the shader. It's a little bit different in that it's using it to modify how a point cloud is drawn instead of a 2D image, but good for helping to understand the technique:
https://github.com/elliotwoods/ofxKinectForWindows2/tree/master/exampleBodyIndexShader

Also, you should try posting in the [OF forums](https://forum.openframeworks.cc/)!

rwebber, 2017-01-13:
Much appreciated...
I am still slowly working on this.
You can take a peek here:
https://github.com/rwebber/kinect2share

I will start using the OF forum.. I see you are rather active there.
Cheers

elliotwoods, 2017-01-14:
Nice!
Also you might want to check out ofxMultiTrack for another implementation
of low latency network streaming full Kinect data

closed 2021-04-03 by elliotwoods