elliotwoods/ofxKinectForWindows2 #67: reelase version errors

bluelemonade, 2016-10-26:
I tried to compile a release version with the ofxkinectforwindows2, the debug version runs!

I got: Cannot open include file: 'Kinect.h': No such file or directory (compiling source file ......\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Data\Body.cpp)	test_ofxkinect2forwindows e:\of_v0.9.0_vs_release\addons\ofxkinectforwindows2\src\ofxkinectforwindows2\data\Joint.h	8

should I link something in the project properties in vs2015?

elliotwoods, 2016-10-26:
Kinect.h comes from the official SDK
By adding the props file to your project (as explained in readme) this
should be added to your include path
If you haven't rebooted then please try
Reboot makes sure that your system paths are set correctly after any
changes are made (e.g. Install SDK)

bluelemonade, 2016-10-26:
ok, I added the props file, where can I see that it down right?

when adding the reference I can't select the lib? theres only the openframework lib...

elliotwoods, 2016-10-27:
Please paste screenshots where possible

You can see if the property sheet is correctly applied in the property manager 

I suggest checking solution properties to make sure that all the build settings are right for each included project (debug/release x86/x64

bluelemonade, 2016-10-27:
hi elliot,

the lib is now included, but already having still error...

regards

[image]

elliotwoods, 2016-10-31:
it seems like you've missed a step from the readme
https://github.com/elliotwoods/ofxKinectForWindows2#usage

please make sure that the ofxKinectForWindows2 project is inside your solution (step 1 in readme):
[image]

you seem to have step 2

and for step 3 please check:
[image]

if the reference is missing, please try:
[image]

bluelemonade, 2016-10-31:
ooops, I missed step one! i am a f...

elliotwoods, 2016-11-01:
a "f"?? 
anyway, seems resolved so closing.
good luck!

closed 2016-11-01 by elliotwoods