elliotwoods/ofxKinectForWindows2 #68: use kinect.getDepthSource()->getPixels() to color foreground and background with two different colors

fdiba, 2016-10-31:
Hello, I try to make sense of the examples by using kinect.getDepthSource()->getPixels() to color foreground and background with two different colors.

In ofApp::update I have written this :

`for (int y = 0; y < DEPTH_HEIGHT; y++) {

			for (int x = 0; x < DEPTH_WIDTH; x++) {

				int index = x + y*DEPTH_WIDTH;

				if (depthPix[index] < 1000) {

					ofColor c = ofColor::red;
					img.setColor(x, y, c);

				} else {

					ofColor c = ofColor::black;
					img.setColor(x, y, c);

				}

			}

		}

		img.update();

	}`

But I get always the same error "ofPixels_<unsigned short>::operator[](...) returned nullptr." when I use "depthPix[index] < 1000".

Any tips ?

elliotwoods, 2016-11-01:
what's the value of `index` when you get the runtime error?

did you check if there is a valid frame before attempting to process it? e.g.:

```c++
auto depthSource = kinect.getDepthSource();
if(depthSource->isFrameNew()) {
    auto depthPixels = depthSource->getPixels();
....
}
```

It might be simply that you're trying to process the first frame before it ever arrives.

fdiba, 2016-11-01:
Thanks! I should have put a closer look to this condition, present in the examples :

`if (!depthPix.size() || !bodyIndexPix.size() || !colorPix.size())`

Works fine!

closed 2016-11-01 by fdiba