elliotwoods/ofxKinectForWindows2 #69: does this framework work with two cameras?

LauraLaureus, 2016-12-07:
The question in the title, does this of add on work with several cameras, I need to work with two for my master project.

mattfelsen, 2016-12-07:
Hey @LauraLaureus. This addon is a wrapper around Microsoft's official Kinect 2 SDK, which only supports one camera per computer. A couple options would be to have multiple computers and send data from one to the other over a network, or to use a different library such as [libfreenect2](https://github.com/OpenKinect/libfreenect2) with multiple camera support, though it requires a special PC with multiple USB cards (specifically you need multiple USB busses). Also that library only provides access to raw data streams (depth, color, IR, etc.) and no higher-level APIs like skeleton-, hand-, or face-tracking, so that may impact your decision as well. Good luck!

LauraLaureus, 2016-12-07:
Okey, thank you @mattfelsen for the info. ^^

closed 2019-11-14 by elliotwoods