elliotwoods/ofxKinectForWindows2 #70: Access to the 3d Pointcloud

iamterryclark, 2017-01-08:
Hi Elliot,

This library is great thanks for making it.

I wondered if you can help me understand how I might access the pointcloud through 3d Vector I want to use the pointcloud points as origins of a particle system so I need to threshold the pointcloud to only display a certain z axis point and also only display every nth pixel in the pointcloud.

Many thanks in advance.

Terry

elliotwoods, 2017-01-08:
Use worldSource->getMesh().getVertices() to get a vector of vertices in
world space

closed 2017-03-08 by elliotwoods

CharStiles, 2019-11-09:
Hi, 
I was wondering what you mean by worldspace? I'm wondering because when I try to get the mesh nothing returns.
`
ofxKFW2::Device kinect;
kinect.open();
kinect.initDepthSource();
ofMesh scanFull = kinect.getDepthSource()->getMesh();
`
Is nothing coming out because I'm not in worldspace?
Thanks a lot for writing this addon!