elliotwoods/ofxKinectForWindows2 #72: CL.exe could not be run.

bakercp, 2017-01-20:
Hey @elliotwoods, I hate to bother you with this one, as I'm almost certain it's a configuration problem on my end, and my lack of Windows experience is on show ... but hoping you have a tip:

[image]

Have you seen this?  Online it usually says that this means that C++ support hasn't been installed, but I made sure it was during installation and confirmed other oF examples compile no problem.

Anyway, everything was working splendidly on a different windows machine yesterday.  Then I moved to a fresh install of Windows 10, installed VS2015 community, the Kinect SDK 2.0, etc and started getting this error, just with this addon.  I can compile openFrameworks apps (examples, etc) normally.

Let me know if you have thoughts or would prefer for me to take this to the forum.

Thanks,
CB

elliotwoods, 2017-01-20:
Can you try changing your of path to something shorter?
E.g. C:\of
Just to test if that's the problem?
One of the vs build apps can't handle commands over 255 chars in length

bakercp, 2017-01-20:
@elliotwoods Wow.  That was it.  You just saved me a whole lot of time.  I'll add a little section to the README.md as my path wasn't insanely unusual, but apparently the `ofxKinectForWindows2Lib` sub project path length put it over the edge.  Thanks again.

closed 2017-01-20 by bakercp