elliotwoods/ofxKinectForWindows2 #76: Body::getFloorTransform() inaccurate?

tyhenry, 2017-02-14:
I can't seem to draw the floor plane accurately, either manually drawing it myself using the floor transform matrix or by using the drawWorld() method.

The floor transform matrix is definitely shifted from the identity matrix, but clearly not compensating for the angle of my Kinect (which is about 7 ft high on a shelf and fully tilted ~35 deg?).  The plane also bisects my 3D scene rather than sitting at the floor.  

KinectStudio shows the correct floor in 3D mode so it's not the SDK's floorClipPlane values that are the issue.  I'm guessing it's the math in Body::getFloorTransform(), but it's been a while since I took a math class...

However, I found this great step by step guide (stack exchange answer) for using the floorClipPlane to create a matrix (where origin is closest point on plane to Kinect, e.g. directly under the device on floor): 
http://gamedev.stackexchange.com/a/80489

and the Wolfram plane page is somewhat helpful too http://mathworld.wolfram.com/Plane.html

just wondering if there's a quicker fix for this I'm missing

tyhenry, 2017-02-15:
Here's a quick fix that's working for me.
in Body.cpp change Body::getFloorTransform() to:

ofxMatrix4x4 Body::getFloorTransform() {
    ofVec3f normal(floorClipPlane.x,floorClipPlane.y,floorClipPlane.z); // floor normal
    float distance = floorClipPlane.w;  // floor origin distance from (0,0,0)
    ofMatrix4x4 rot, trans;
    rot.makeRotationMatrix(ofVec3f(0, 1, 0), ofVec3f(normal)); // rotation of plane
    trans.makeTranslationMatrix(normal*-distance); // origin of plane
    return rot*trans;
}

will submit a pull request tomorrow with the update

elliotwoods, 2017-02-15:
@tyhenry - thank you for researching this!
i presume you're using the non-glm version here then?
so making a PR is a little tricky.
i'll leave this open and implement in glm when i'm back on PC and checking issues

tyhenry, 2017-02-15:
Oh yeah, good point!  I'm on 0.9.8... haven't switched to glm yet

tyhenry, 2017-02-15:
if you want to create a new branch from the 0.9.0 tag, I could PR that one for the legacy syntax