elliotwoods/ofxKinectForWindows2 #78: Tracking Hand State

carloscastellanos, 2017-03-16:
I'm having trouble tracking hand states and I just wanted to make sure whether it's my code the library or the Kinect itself.

Joint positions seem to be tracking fine but for some reason the only response I get from the HandState is 0 (unknown) or 1 (not tracked). I see no reason why this code shouldn't work. Am I doing something wrong?

Here is my code:
`
	float distLeft = 0.0;
	float distRight = 0.0;
	float handLeftState = 0;
	float handRightState = 0;

	// ==== get the Kinect data first ==== //
	kinect.update();

	// Get joint positions (skeleton tracking)
	{
		auto bodies = kinect.getBodySource()->getBodies();
		unsigned int trackCount = 0;
		for (auto body : bodies) {
			if (body.tracked) { // make sure the body is actually being tracked
				trackCount++; 
				if (trackCount > 0 && trackCount <= MAX_BODY_TRACK) {
					auto joints = body.joints;
					// get hand positions
					float handLeftX = joints[JointType_HandLeft].getPosition().x;
					float handLeftY = joints[JointType_HandLeft].getPosition().y;
					float handLeftZ = joints[JointType_HandLeft].getPosition().z;
					float handRightX = joints[JointType_HandRight].getPosition().x;
					float handRightY = joints[JointType_HandRight].getPosition().y;
					float handRightZ = joints[JointType_HandRight].getPosition().z;
					// get hand states (open=2, closed=3, lasso=4, notracked=1, unknown=0)
					handLeftState = body.leftHandState;
					handRightState = body.rightHandState;
					//ofLog(OF_LOG_NOTICE, "left hand state: " + ofToString(handLeftState));
					// get head position
					float headX = joints[JointType_Head].getPosition().x;
					float headY = joints[JointType_Head].getPosition().y;
					float headZ = joints[JointType_Head].getPosition().z;
					// get eucidian distance between hands and head
					distLeft = ofDist(handLeftX, handLeftY, handLeftZ, headX, headY, headZ);
					distRight = ofDist(handRightX, handRightY, handRightZ, headX, headY, headZ);
				}
			}
		}
		// update the global variable numTracked with the number of bodies just tracked
		numTracked = trackCount;
	}`

tyhenry, 2017-03-16:
Is there a reason you declare the hand states as floats?
I would try:

```
HandState handLeftState, handRightState;     // not float
...

for (auto& body : bodies) {    // use reference & to avoid copy
     ...

     handLeftState = body.leftHandState;
     ofLogNotice() << "left hand: " << handLeftState; 
```

Also, you're overwriting your hand state values for each body, not sure what the aim is here.

carloscastellanos, 2017-03-16:
Yeah I have the hand states as floats because I send them to a data structure that requires floats. I had them as ints before and had the same issue.

My data structure was outside of the for loop so it was being overwritten before being stored as you said. Dumb mistake (I've been testing using only 1 body so overwriting hand state values shouldn't be any issue anyway). I fixed that (and am using reference as you suggested) but it is still not recognizing hand states.

tyhenry, 2017-03-16:
Hmm not sure then -
I have a [(hacky) addon that will draw the hand states](https://github.com/tyhenry/ofxKinect2User/tree/master/src), if you want to test the Kinect.  Open an issue on my addon if you want some info on how to use it (basically create a "User", update it with a body from the Kinect, and draw the joints).

but just to rule out the enum-float conversion (which shouldn't be an issue, but...) I'd suggest something like:

```
HandState handLeftState, handRightState;
...

handLeftState = body.leftHandState;

...
vector<float> myDataStructure;
float leftF = (int)handLeftState;
ofLogNotice() << leftF;
myDataStructure.push_back(leftF);
```

carloscastellanos, 2017-03-17:
Thanks, I'll fiddle around with it some more and try your code and/or your addon.

carloscastellanos, 2017-03-17:
OK, I figured it out! It seems that the issue was due to the gpu driver. I update the driver for my Nvidia card and hand states work now. The issue is actually addressed on the [Microsoft forums](https://social.msdn.microsoft.com/Forums/en-US/dd14d569-a03c-4abc-b740-f7f920746e30/handstate-error-in-1408-1409-sdk-please-check?forum=kinectv2sdk)

Feel free to close this issue.

closed 2021-04-03 by elliotwoods