elliotwoods/ofxKinectForWindows2 #79: Setting up failed, ofRotateDeg() and glm not found

Aerobraking, 2017-03-25:
Hey guys,

I'm just starting with openframeworks and ofxKinectForWindows2, but directly failed to set it up. My Kinect and openframeworks are working fine. When I add the ofxKinectForWindows2 and try to compile any of the examples in visual studio 2105, I get at first the error that in the Device.cpp, the method ofRotateDeg() in Line 370. is undefined. If I comment that out, the next error is:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C3861	'inverse': identifier not found	ofxKinectForWindows2Lib	c:\of_v0.9.8_vs_release\addons\ofxkinectforwindows2\src\ofxkinectforwindows2\source\body.cpp	21
Error (active)		identifier "ofRotateDeg" is undefined	ofxKinectForWindows2Lib	c:\of_v0.9.8_vs_release\addons\ofxKinectForWindows2\src\ofxKinectForWindows2\Device.cpp	370	
Error	C2653	'glm': is not a class or namespace name	ofxKinectForWindows2Lib	c:\of_v0.9.8_vs_release\addons\ofxkinectforwindows2\src\ofxkinectforwindows2\source\body.cpp	21
```	

I tried it with of version 0.9.0 and 0.9.8, both have the same problems. 

I guess it is something stupid that I am overseeing, so sorry if the answer is obvious but I don't see it.

Cheers
Konnie

mattfelsen, 2017-03-25:
Take a look at https://github.com/elliotwoods/ofxKinectForWindows2/issues/77. You'll to use the `0.9.0` tag, not the `master` branch, of this addon:
https://github.com/elliotwoods/ofxKinectForWindows2/tree/0.9.0

Aerobraking, 2017-03-25:
Thank you! I forgot to have a look at the closed issues and only looked at the open ones, otherwise I would found that for myself. :D 

It is working now. :)

mattfelsen, 2017-03-25:
No worries, glad you're up & running. I just submitted a PR to update the readme so it's easier to catch for folks getting started: https://github.com/elliotwoods/ofxKinectForWindows2/pull/80

closed 2021-04-03 by elliotwoods