elliotwoods/ofxKinectForWindows2 #82: Infrared Image compressed?

Aerobraking, 2017-10-18:
Hey guys,

I'm using the infrared source in my app. As the raw image is too dark, I brighten it up. But in comparison to the infrared image I get from the Kinect Studio or the example projects from the Kinect SDK, the image quality is lower.

Here is a side by side (left Kinect studio, right ofxKinectForWindows2 ), both images are brightened in photoshop. I already looked at the pixel data from the infraredsource, the original pixel are already at this "low" quality.
[image]

Any ideas why and how I could maybe get the higher quality?

Cheers
Konnie

elliotwoods, 2019-11-14:
Hi Konnie
oh wow that's really weird

i'm sorry for replying 2 years too late also haha!

i presume it's something to do with the way that the data is drawn in oF
i.e. you might be using uint8 data structures rather than uint16 in your app?
e.g. if you were to put the data into a uint8 and then multiply, then you'd see the type of gradient stepping that you see in your right hand image. If you multiply in a uint16 and then copy to uint8 you should see smooth gradients like on the left.

Aerobraking, 2019-11-17:
Hey elliot, thanks for reply. Bate late then never. :D 

Yes, I think that was the mistake I did back then.

closed 2021-04-03 by elliotwoods