elliotwoods/ofxKinectForWindows2 #83: exampleDepthToCameraTable

fred-dev, 2017-11-10:
exampleDepthToCameraTable is impossible to compile, it relies in many add ons, and ofxassets will not work as it relies on poco that is no longer part of OF. It would be great to be able to learn more from these examples, especially if they used more standard OF stuff. It is a bit tough to see what is going on.

elliotwoods, 2017-11-12:
hey!
all you need to do is in the update loop
and the update loop doesn't interact with the gui at all

https://github.com/elliotwoods/ofxKinectForWindows2/blob/master/exampleDepthToCameraTable/src/ofApp.cpp#L76

elliotwoods, 2017-11-12:
Sorry if you spent a while trying to get this to build and the addons weren't working at the time for you.
I don't have oF 0.10 tagged versions of my addons yet as i haven't upgraded myself (the changes to math libraries require much work to update everything ive done so far, and i haven't had time yet to look into it)

closed 2017-11-12 by elliotwoods