elliotwoods/ofxKinectForWindows2 #84: Error in setup addon in visual studio community 2017

Goodvirus, 2017-12-28:
Hello,
     I downloaded openframework version 0.9.8 and also  ofxKinectForWindows2 addon 0.9.0. I also did all steps mentioned in Readme.md file. Still getting following error. Is anybody know how to solve them?

[image]

Thanks in advance

elliotwoods, 2017-12-28:
Heya!
Quick question :
Does the file which it is complaining about exist?

Goodvirus, 2017-12-28:
Hello Elliotwood,
       Thanks for reply, I didnt find any file ofxKinectForWindowsLib.lib in addon folder. I don't know why this error is occured.

elliotwoods, 2017-12-28:
https://github.com/elliotwoods/ofxKinectForWindows2/blob/0.9.0/src/ofxKinectForWindows2/Data/Joint.h#L3

This is the file it's complaining about
please can you check if it exists

the lib file is generated by your linker once the addon has been successfully compiled

Goodvirus, 2017-12-28:
yes the file exists within the project
[image]

elliotwoods, 2017-12-28:
Does the file actually exist though?
e.g. if you click on it in the solution explorer, then does it open up
and can you find it in the filesystem?

Goodvirus, 2017-12-28:
yes file is physically exist at specified directory
[image]
[image]

Goodvirus, 2017-12-28:
I added the file directory into including directory of project properties and change command in Utils.h from #include "ofKinectForWindows/Data/Joint.h" to #include <Joint.h> . It solves my problem

Thanks for help @elliotwoods

armonnaeini, 2019-04-12:
Hi I have the exact same errors as you do, and I have no idea how to solve this. Please someone help me out, I have been struggling with this for too long. When you say you
"added the file directory into including directory of project properties" what do you mean by that??

What is the including directory of project properties? How can I access that?

elliotwoods, 2019-04-12:
@Goodvirus @armonnaeini - I'm sorry but I'm not familiar with this error fo what could be causing it.

@armonnaeini - Goodvirus is suggesting that you add the folder containing Joint.h directly into your project properties include path by:

1. right click on your Project (e.g. 'emptyExample') in the solution explorer and select 'Properties'
2. On the left navigate to 'C/C++/General'
3. Look for the top entry - 'Additional Include Directories'
4. Click on the text data in the right hand cell, click the down arrow, click Edit. You should get a pop up
5. Add a new entry - the directory which contains Joint.h
6. Edit Utils.h as mentioned by Goodvirus above

armonnaeini, 2019-04-12:
@elliotwoods 

Thank you so much for your help. That resolved that respective error. I am now getting another error here:

"cannot open file 'C:\Users.......\exampleBodyIndexColor\bin\ofxKinectForWindows2Libd.lib'"

Would you happen to know why I would be receiving this error? Should I take the same steps as before, enter the direct path to the .lib file in the directories panel, within my solution properties?

I guess I would just like to see what is the most viable solution whenever I run into these kinds of petty linking errors. Thanks so much for your help, I seriously appreciate it.

elliotwoods, 2019-04-12:
if it can't find that lib file, that generally means that it didn't build the addon correctly
try
1) Make sure there is a reference to ofxKinectForWindow2 on your app project, look like:
[image]

(should also be a reference to openFrameworks too)

2) Try a rebuild solution

armonnaeini, 2019-04-12:
That's strange, both of those references are present in my solution explorer pane...I don't know what the problem could be here...

elliotwoods, 2021-04-03:
Ah sorry @Goodvirus - from looking at the screenshots it seems that you're not using the standard folder structure.
That isn't supported by ofxKinectForWindows2Lib (this is mentioned in the readme - maybe could be more explicit)

closed 2021-04-03 by elliotwoods