elliotwoods/ofxKinectForWindows2 #85: 'toggleCursorDrawEnabled': is not a member of 'ofEasyCam'

amygoodchild, 2018-01-21:
I'm getting this error in a couple of the examples, such as "exampleWithGui" and "exampleBodyIndexShader".

```
1>src\ofApp.cpp(88): error C2039: 'toggleCursorDrawEnabled': is not a member of 'ofEasyCam'
1>C:\openFrameworks\libs\openFrameworks\3d\ofEasyCam.h(8): note: see declaration of 'ofEasyCam'
1>src\ofApp.cpp(88): error C2039: '__this': is not a member of 'ofEasyCam'
1>C:\openFrameworks\libs\openFrameworks\3d\ofEasyCam.h(8): note: see declaration of 'ofEasyCam'

```
This is the bit of code in ofApp.cpp:

```
//if we press the 'c' key on the World panel, then toggle the camera's cursor. This works best when you fullscreen that panel
	worldView->onKeyboard += [this, worldView] (ofxCvGui::KeyboardArguments & args) {
		if (args.action == ofxCvGui::KeyboardArguments::Action::Pressed && args.key =='c') {
			worldView->getCamera().toggleCursorDrawEnabled();
		}
	};
```
I have ofxCvGui and all the associated addons it asked for linked.

Any ideas on this?

elliotwoods, 2018-01-21:
Hi Amy!

I’m sorry that toggleShowWorldCursor is from ofxGrabCam.
You can just remove the lines of code relating to the ‘c’ key if you are using ofEasyCam

amygoodchild, 2018-01-22:
Hey Elliot! Thanks so much for the speedy reply. 

I did try adding ofxGrabCam to my addons but it doesn't seem to make a difference. I also tried commenting out that section but then I get the following unresolved external symbol errors.

These are the errors:
```
1>TextField.obj : error LNK2019: unresolved external symbol "public: void __thiscall ofxCvGui::Utils::TextField::defocus(void)" (?defocus@TextField@Utils@ofxCvGui@@QAEXXZ) referenced in function "public: void __thiscall <lambda_9f2538dad2349565f98790bcf0a4fa11>::operator()(class ofxCvGui::MouseArguments &)const " (??R<lambda_9f2538dad2349565f98790bcf0a4fa11>@@QBEXAAVMouseArguments@ofxCvGui@@@Z)
1>TextField.obj : error LNK2019: unresolved external symbol "public: bool __thiscall ofxCvGui::Utils::TextField::isFocused(void)const " (?isFocused@TextField@Utils@ofxCvGui@@QBE_NXZ) referenced in function "public: void __thiscall <lambda_9f2538dad2349565f98790bcf0a4fa11>::operator()(class ofxCvGui::MouseArguments &)const " (??R<lambda_9f2538dad2349565f98790bcf0a4fa11>@@QBEXAAVMouseArguments@ofxCvGui@@@Z)
1>bin\exampleWithGui_debug.exe : fatal error LNK1120: 22 unresolved externals
```

Here's an excerpt of how it looks in the error list:
[image]

I am not sure if this is related to having removed that section of code or just the next issue on the list. I had some similar errors before but solved them by making sure the link to the ofxKinectForWindows2Lib project and the reference was set up correctly. Is there something else I need to link? Thanks again!

elliotwoods, 2018-01-22:
It looks like ofxTextInputField.cpp is missing from the project. Please can you add it somewhere? It's seeing the header, but the symbols are not being built

amygoodchild, 2018-01-22:
Hmm, it seems to be there...

[image]

And the file exists...

[image]

elliotwoods, 2018-01-24:
hey!
on my laptop now so can seethe screenshot properly :)

please can you try adding ofxAssets to your project?
https://github.com/elliotwoods/ofxAssets

amygoodchild, 2018-01-24:
Hey, thanks again for the help :)

ofxAssets is there already though...

[image]

elliotwoods, 2018-07-18:
exampleWithGui builds here for me as it is in the GitHub repo.
I've tested against the current master versions of:

* ofxAssets
* ofxCvGui
* ofxGrabCam
* ofxLiquidEvent
* ofxSingleton
* ofxTextInputField (https://github.com/elliotwoods/ofxTextInputField not Flightphase)
* ofxKinectForWindows2 (tag 0.9.0)

Could you test with those?
I've also zipped up all these addons for you at:
https://drive.google.com/open?id=1B7M743pLmlsq0Rqq0dToie2MZGygyDsp

closed 2019-11-14 by elliotwoods