elliotwoods/ofxKinectForWindows2 #87: Unable to compile version of_v0.10.1_vs2017release

Aurelian24, 2019-04-04:
I´ve been using this library for two years now. It works in of_v0.9.8_vs_release version but when i try to compile it in  of_v0.10.1_vs2017_release it doesn´t work. 

This is the error that im getting , what seems to be the problem? where is that file supposed to be? 

What´s going on?

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'libboost_filesystem-vc140-mt-1_64.lib'	ShaderVisualizer	D:\of_v0.10.1_vs2017_release\apps\myApps\ShaderVisualizer\LINK	1

Aurelian24, 2019-04-09:
UPDATE : 

It seems like the error is when you try to use this library in combination with ofxGuiExtended,

I´ve removed the addon from my solution and the solution compiles !

elliotwoods, 2019-04-10:
Hey there!
Oh weird, as far as I can find, I'm not using any file functions in ofxKinectForWindows2, so not sure where the conflict with libboost's file functions (which are the draft `std` file functions for >C++14)

I'll close this for now, unless we find out that it's an issue with ofxKFW2 itself

Thank you

closed 2019-04-10 by elliotwoods

lutzer, 2019-05-14:
i got the same error, when i tried to compile the examples in Visual Studio Community 2017 without upgrading the project to the new Toolset. When opening the Project, simply confirm to upgrade the upgrade and it will compile.

elliotwoods, 2019-05-14:
thanks @lutzer !