elliotwoods/ofxKinectForWindows2 #88: Head joint orientation

natxopedreira, 2019-06-11:
The head joint orientation always return 0 however other joints orientation works ok.

ofQuaternion orNeck = body.joints[JointType_Neck].getOrientation(); // works

ofQuaternion orHead = body.joints[JointType_Head].getOrientation(); // 0 value

elliotwoods, 2019-06-11:
this is the data coming from the SDK

https://social.msdn.microsoft.com/Forums/expression/en-US/34542d98-4ed8-44e1-a52d-52c6b39bc74d/why-are-orientations-of-head-and-feet-0000?forum=kinectv2sdk

this is because the head is a leaf node in the skeleton (only has 1 connection)

sorry but nothing i can fix from my side

closed 2019-06-11 by elliotwoods

natxopedreira, 2019-06-11:
I see, it makes sense. I will take a look at face tracking system
Thank you!