elliotwoods/ofxKinectForWindows2 #89: kinect.getDepthSource()->getMesh() returns plain plane

CharStiles, 2019-11-14:
Hi, thanks for making this addon! Its wonderful to have access to kinect on windows in OF. 
I was wondering why these lines return a blank mesh, i.e. a plane.
`ofxKFW2::Device kinect; 
kinect.open(); 
kinect.initDepthSource(); 
ofMesh scanFull = kinect.getDepthSource()->getMesh();`
I am using VS 2017 with a kinect v2. The IR returns a texture fine, I haven't been using the depth texture because it lacks details. its either the same gray color or black, no inbetween values. I'm not sure if thats relevant i can provide more details if needed! I am invested in the future of this addon.

elliotwoods, 2019-11-14:
hey!
that should work fine
are you calling `kinect.update()` each frame also?

elliotwoods, 2019-11-14:
Also note that the first few frames are often blank whilst the device initialises (so you can't test for a valid mesh until you've been calling kinect.update() for a couple of seconds

closed 2021-04-03 by elliotwoods